jfjlaros/simpleRPC #2: Graceful disconnection & const char to PROGMEM

apfeiffe, 2019-02-10:
Hi Jeroen:

**Is your feature request related to a problem? Please describe.**
- Python:It would be nice to terminate a connection between python program and Arduino gracefully to change serial connection parameter without stopping the python program
- Arduino: Please check if the const char[] of the interface() definition can be moved to the flash memory (PROGRAM) iso SRAM (DATA). At the moment I'm running low while defining a number of interface() routines, e.g.
```
DATA:    [=======   ]  70.7% (used 1448 bytes from 2048 bytes)
PROGRAM: [===       ]  30.8% (used 9472 bytes from 30720 bytes)
```
**Describe the solution you'd like**
- Python: something like:
```
interface = Interface('/dev/ttyACM0')
interface.connect()
interface.terminte()
```
- Arduino:
Using `F() macro and PROGMEM`

Thanks,
Andy

jfjlaros, 2019-02-10:
Thank your for the suggestions.

I have added an optional `autoconnect` parameter to the constructor and the functions `open()`, `close()` and `is_open()`. I have also added support for the `with` syntax to open and close automatically.

Could you test the Python library to see whether it fits your needs?

jfjlaros, 2019-02-13:
I added support for optional use of the `F()` macro in the `interface()` function. It seems to work as intended.

apfeiffe, 2019-02-14:
Alright. I did some quick initial test on both changes (Python and Arduino) and I'm satisfied. So far no regressions were seen.

Here are some numbers to compare:

- Arduino:
```
DATA:    [==        ]  23.0% (used 472 bytes from 2048 bytes)
PROGRAM: [===       ]  30.9% (used 9504 bytes from 30720 bytes)
```
If there's anything left open, I'll let you know.

Many thanks, Jeroen!

closed 2019-02-14 by apfeiffe

jfjlaros, 2019-02-15:
I have a question. Where do you get the memory usage statistics from? I used the [MemoryFree](https://github.com/maniacbug/MemoryFree) library for testing the PROGMEM functionality, but you seem to have a better tool for that.

apfeiffe, 2019-02-15:
I've been using Visual Studio Code together with PlatformIO ([https://platformio.org/](url)) within an Ubuntu VM.

jfjlaros, 2019-02-15:
Ah, I see. Thanks.

jfjlaros, 2019-03-05:
Just to avoid confusion. In case you are wondering where the Python code is, I have extracted the client and put it in its own repository [here](https://github.com/jfjlaros/arduino-simple-rpc).