jfjlaros/simpleRPC #3: Heap corruption with Vector's as function arguments

iot49, 2019-06-28:
I'm getting heap corruption when passing Vector arguments to functions called by simpleRPC.

Setting Vector.destroy = false in function void _read(Vector <T> *data) in read.tcc appears to fix the problem, although this may (will?) result in a memory leak.

Test code (snippet):

```
Vector<uint8_t> send_Vector(Vector<uint8_t> data) {
    ESP_LOGD("esp_connect", "send_Vector %d", data.size);
    return data[1];
}

interface(send_Vector, F("send_Vector"));
```

jfjlaros, 2019-06-28:
This is probably because the vector is made (and should be destroyed) outside of the function `send_Vector()`. By passing the vector as is, the destructor is called when the function ends, the calling function will additionally call the destructor, which leads to undefined behaviour.

My guess is that by passing the vector by reference solves the problem, i.e.,

    Vector <uint8_t>send_Vector(Vector <uint8_t>&data) { ... }

Does this help?

jfjlaros, 2019-06-28:
By the way, your return type (Vector) does not seem to match the type of the value you return (uint8_t).

There are some examples on complex data structures in the online [example](https://github.com/jfjlaros/simpleRPC/blob/master/examples/demo/demo.ino). Perhaps that helps.

iot49, 2019-06-28:
Yes, passing a reference solved this problem. Many thanks!

Other than that I had no issues - with simpleRPC, that is ...

closed 2019-06-28 by iot49

jfjlaros, 2019-06-28:
Thanks for the feedback.