jfjlaros/simpleRPC #5: int vec_test3(Vector<uint8_t> &v, int flags) --> CRASH

iot49, 2019-07-19:
Calling the following function via RPC results in a crash (corrupt heap detected) when the function returns.

```C++
int vec_test3(Vector<uint8_t> &v, int flags)
{
    return flags+1;
}
```

Without an `int` after the `Vector`, there is no issue. 

```c++
Vector<uint8_t> vec_test1(Vector<uint8_t> &v)
{
    Vector<uint8_t> res = Vector<uint8_t>(v.size);
    for (int i=0;  i<v.size;  i++)
        res[v.size-i-1] = v[i];
    return res; 
}

Vector<uint8_t> vec_test2(int sz)
{
    auto res = Vector<uint8_t>(sz);
    for (int i=0;  i<sz;  i++) res[i] = 'A' + i;
    return res;
}
```

jfjlaros, 2019-07-20:
Can you try it now?

closed 2019-07-20 by jfjlaros

bboser, 2019-07-20:
Wow, thanks for the quick fix! Yes, no more problems.

With hindsight I see why this should help, but I never would have found the
problem.

Thanks,
Bernhard

jfjlaros, 2019-07-20:
I got quite some help from Valgrind, which reported an invalid `free()`. This suggests there was a problem like the one we have seen in #3.

Thank you for raising this issue.

bboser, 2019-07-23:
Hi Jeroen,

How do you use Valgrind with an Arduino?

Do you test it on a host? Would be nice to be able to check memory leaks on
the esp32.

Regards,
Bernhard

jfjlaros, 2019-07-23:
Well, I cheat a bit I suppose.

I wrote a small [fixture](https://github.com/jfjlaros/arduino-serial-fixture) library to simulate a serial interface. This library is used in the [unit tests](https://github.com/jfjlaros/simpleRPC/tree/master/tests) which in turn use the [Catch2](https://github.com/catchorg/Catch2) unit testing framework. These unit tests are then [executed](https://github.com/jfjlaros/simpleRPC/blob/master/tests/Makefile) using Valgrind.

This approach is not 100% full proof, for one there are slight differences between the C++ version of Arduino and C++11 (which is enforced for the tests). It does allow for easy [CI](https://travis-ci.org/jfjlaros/simpleRPC) though.

iot49, 2019-07-23:
Thanks, I have some serious learning to do.

Bernhard

commit d1885f4f (2019-07-20): Bug fix #5: parameters not passed by reference.
commit 8cafe738 (2019-07-20): Bug fix #5: parameters not passed by reference.