jfjlaros/simpleRPC #6: Compiling Error with CONTROLLINO MEGA

gubely, 2020-07-30:
**Describe the bug**
We are using simpleRPC on a Controllino MEGA which has worked great so far. Since we updated your library from 2.0.1 to 3.0.0, we receive an compiling error.

**To Reproduce**
We run this code on our Controllino:

`   #include <Controllino.h>   Usage of CONTROLLINO library allows you to use CONTROLLINO_xx aliases in your sketch.
    #include <simpleRPC.h>
    
    int i=0;
    const int lowestInputPin = 0;
    const int highestInputPin = 15;
    const int lowestOutputPin = 0;
    const int highestOutputPin = 16;
    int CONTROLLINO_D = 0;
    int CONTROLLINO_A = 0;
    byte pump_speed[11];
    byte flow_raw[11];

byte get_byte_element(byte index) {
  return flow_raw[index];
}

bool set_byte_element(byte index, byte value) {
  pump_speed[index] = value;
  return true;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);//

  // set Inputpins 0 - 15 as analog Inputs:
  for (int thisInputPin = lowestInputPin; thisInputPin <= highestInputPin; thisInputPin++) {
    pinMode(CONTROLLINO_A+thisInputPin, INPUT);
    }
  // set Outputpins 0 - 16 as digital outputs:
  for (int thisOutputPin = lowestOutputPin; thisOutputPin <= highestOutputPin; thisOutputPin++) {
    pinMode(CONTROLLINO_D+thisOutputPin, OUTPUT);
    }
  int  PumpspeedAblauf = 200;
  int  PumpspeedFilter = 150;
  analogWrite(CONTROLLINO_D11, PumpspeedAblauf);                        // abpumpen
  analogWrite(CONTROLLINO_D16, PumpspeedFilter);                        // Filtern
}

void loop() {
  interface(
  digitalRead,
    "digital_read: Read digital pin. @pin: Pin number. @return: Pin value.",
  digitalWrite,
    "digital_write: Write to a digital pin. @pin: Pin number. @value: Pin value.",
  analogRead,
    "analog_read: Read analog pin. @pin: Pin number. @return: Analog value.",
  analogWrite,
    "analog_write: Write to an analog pin (takes 1.2s to proceed). @pin: Pin number. @value: Analog value.",
  get_byte_element,
    "get_byte_element: Reads a byte element from array. @index: index of element. @return: element.",
  set_byte_element,
    "set_byte_element: Writes a byte element into an array. @index: index of element. @value: value of element. @return: acknowledgement.");

  for (i=0; i<=10; i++){
    analogWrite(CONTROLLINO_D0+i, pump_speed[i]);
    flow_raw[i] = analogRead(CONTROLLINO_A0+i);
  }
}`

(refer to [this](https://www.controllino.biz/tutorials/#1539865447635-8a783d0e-bd22) library setup)

and as we compile this error code occurs:

`Arduino: 1.8.13 (Windows 10), Board: "CONTROLLINO MEGA"

In file included from C:\Program Files (x86)\Arduino\libraries\simpleRPC\src/simpleRPC.h:5:0,

                 from ...\arduino\port_extender\port_extender_Pressure\port_extender_Pressure.ino:2:

C:\Program Files (x86)\Arduino\libraries\simpleRPC\src/interface.tcc: In instantiation of 'void interface(I&, Args ...) [with I = int(unsigned char); Args = {const char*, void (*)(unsigned char, unsigned char), const char*, int (*)(unsigned char), const char*, void (*)(unsigned char, int), const char*, unsigned char (*)(unsigned char), const char*, bool (*)(unsigned char, unsigned char), const char*}]':

...\arduino\port_extender\port_extender_Pressure\port_extender_Pressure.ino:63:139:   required from here

C:\Program Files (x86)\Arduino\libraries\simpleRPC\src/interface.tcc:110:10: error: request for member 'available' in 'io', which is of non-class type 'int(unsigned char)'

   if (io.available()) {

       ~~~^~~~~~~~~

exit status 1

Error compiling for board CONTROLLINO MEGA.
`

**Expected behavior**
The error code stated above appears. As we switch back to version 2.0.1 the IDE compiles without any error and all works fine.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Arduino: 1.8.13

jfjlaros, 2020-07-30:
This is because the latest version has support for plugins (interfaces other than serial) that need to be passed to the `interface()` function.

Could you have a look at the latest [quick start](https://simplerpc.readthedocs.io/en/latest/#quick-start)? Please let me know if anything is unclear.

closed 2020-08-05 by jfjlaros

gubely, 2020-09-04:
Thanks for your instant reply. I haven't managed to check your suggested solution, but makes total sense to me.

Thank you!