jfjlaros/simpleRPC #7: usb_serial_class plugin support

syzygyzer, 2020-11-15:
**Is your feature request related to a problem? Please describe.**
I want to use simpleRPC with a Teensy 3.6. I tried to run `HardwareSerialIO::begin` with the `Serial` object but the compiler tells me `error: no matching function for call to 'HardwareSerialIO::begin(usb_serial_class&)`

**Describe the solution you'd like**
I think a new plugin that works with `usb_serial_class`, or possibly overloading `HardwareSerialIO::begin` to work with it.

**Additional context**
I figured out that this is because the Teensy treats the USB serial port separately from the I/O pin serial ports, unlike most ordinary Arduinos which link the USB TX and RX with the serial TX and RX via a resistor and thus `HardwareSerial` covers both. So with the Teensy they are treated with a separate class, but are still hardware serial supports (at least for the Teensy 3.6, I haven't looked at other ones yet).

I am pretty new to the Teensy/Arduino world, not to mention C++, but this seems like an easy problem that only needs a few lines of code to fix. I will give it the old college try tomorrow.

jfjlaros, 2020-11-15:
I see you have already implemented something. If your solution works, could you submit a pull request?

syzygyzer, 2020-11-16:
I forked the repo and tried my first dumb idea, but it didn't work. Didn't get a chance to work on it yesterday. I ought to have time some night this week though.

jfjlaros, 2020-11-16:
I didn't see any obvious mistakes, except perhaps for a missing prototype in [`io.h`](https://github.com/syzygyzer/simpleRPC/blob/master/src/plugins/hardwareserial/io.h). You might want to add the following line.

    void begin(usb_serial_class&);

jfjlaros, 2020-11-19:
Could you try the last version on the master branch?

chrisflesher, 2020-11-19:
We just merged some code that introduces a `Stream` class, can you try pulling from `master` and see if it works? The syntax would be `HardwareSerialIO::begin(&usb_serial_class)`

syzygyzer, 2020-11-19:
Thanks, I will try this out tonight with my Teensy 3.6 and report back.

syzygyzer, 2020-11-20:
It works! I used `SerialIO::begin` though (which I think is what you meant?). Thank you very much.

closed 2020-11-20 by syzygyzer

jfjlaros, 2020-11-20:
Nice.