jfjlaros/simpleRPC #19: Allow class functions

chrisflesher, 2020-12-06:
It would make things a bit more convenient if it were possible to use class functions:

```
Adafruit_BME280 bme;

void setup(void)
{
  Serial.begin(9600);
  bme.begin();
}

void loop(void)
{
  interface(
    Serial,
    bme.readTemperature, F("get_temperature: @return: Temperature [C]"));
}
```

I'm not sure how difficult / if it is worth doing this with the current template system?

jfjlaros, 2020-12-06:
This is already [possible](https://simplerpc.readthedocs.io/en/latest/usage.html#exporting-class-methods). The following line should work:

    pack(&bme, &Adafruit_BME280::readTemperature), F("get_temperature: @return: Temperature [C]")

chrisflesher, 2020-12-06:
Nice! Thanks.

closed 2020-12-06 by chrisflesher

slaff, 2022-05-02:
And what is solution when this method is overloaded ? For example if `Adafruit_BME280::readTemperature` has the following:
```C++
void Adafruit_BME280::readTemperature(uint8_t& degrees);
uint8_t Adafruit_BME280::readTemperature(); 
```

jfjlaros, 2022-05-02:
To select the first option, something like this should work:

```cpp
(void(Adafruit_BME280::*)(uint8_t&))&Adafruit_BME280::readTemperature
```

slaff, 2022-05-02:
Awesome! Thank you @jfjlaros