jfjlaros/simpleRPC #20: Error HardwareSerial Arduino

joserp93, 2020-12-13:
**Describe the bug**
At the time of compiling the HardwareSerial.ino test project does not compile with the following errors:

In file included from C:\Users\User\Documents\Arduino\libraries\simpleRPC\src/rpcCall.tcc:5:0,

                 from C:\Users\User\Documents\Arduino\libraries\simpleRPC\src/interface.tcc:13,

                 from C:\Users\User\Documents\Arduino\libraries\simpleRPC\src/simpleRPC.h:5,

                 from C:\Users\User\Documents\Arduino\PRUEBARPC\PRUEBARPC.ino:1:

C:\Users\User\Documents\Arduino\libraries\simpleRPC\src/read.tcc: In instantiation of 'void rpcRead(I&, T*) [with I = HardwareSerial; T = unsigned char]':

C:\Users\User\Documents\Arduino\libraries\simpleRPC\src/interface.tcc:113:12:   required from 'void interface(I&, Args ...) [with I = HardwareSerial; Args = {unsigned char (*)(unsigned char), const __FlashStringHelper*}]'

C:\Users\User\Documents\Arduino\PRUEBARPC\PRUEBARPC.ino:18:73:   required from here

C:\Users\User\Documents\Arduino\libraries\simpleRPC\src/read.tcc:19:3: error: no matching function for call to 'HardwareSerial::read(byte*, unsigned int)'

   io.read((byte*)data, sizeof(T));

   ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:232:0,

                 from sketch\PRUEBARPC.ino.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:126:17: note: candidate: virtual int HardwareSerial::read()

     virtual int read(void);

                 ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:126:17: note:   candidate expects 0 arguments, 2 provided

exit status 1
Error compilando para la tarjeta Arduino/Genuino Uno.

jfjlaros, 2020-12-13:
I think you may be using an older version of the library, one that is out of sync with the current documentation. Could you have a look at [this](https://github.com/jfjlaros/simpleRPC/blob/2dfc66310ba39c0d2086bd6cf95b1647d284ffb8/README.rst#quick-start) version of the documentation and try again?

Alternatively, I will release a new version today. Then updating the library should fix this.

jfjlaros, 2020-12-13:
I just released a new version. I expect it to be visible in the Arduino GUI within one day.

jfjlaros, 2020-12-13:
Update: the latest version is now available in the Arduino GUI.

jfjlaros, 2020-12-15:
@joserp93 Can you confirm that any of the two options work?

closed 2020-12-18 by jfjlaros