jfjlaros/simpleRPC #24: simple_rpc: error: invalid protocol header

cdupont, 2021-12-20:
Hello,
thanks for this awesome library! 
I get this error when using simple_rpc:
```
$ simple_rpc call /dev/ttyUSB0 foo 1
usage: simple_rpc [-h] [-v] {list,call} ...
simple_rpc: error: invalid protocol header
```
I suspect it's because I use it with another library that outputs a lot of traces on Serial port.
If I comment out that other library, it works fine.
Any idea how to manage that? I could go and edit the other library, but it's heavy and not under my control...

Thanks

cdupont, 2021-12-20:
Anyway, the question is: how to mix this library and traces? Is it possible?

jfjlaros, 2021-12-20:
I don't think so. A basic serial connection has no notion of multiple channels, there are some higher level protocols that could be used to accomplish this, but if the other library does not conform to that particular standard, you are still out of luck.

Do you perhaps have other communication devices on your board, like a second serial port, Bluetooth or WiFi?

cdupont, 2021-12-20:
No, there is only the USB serial port... Guess I will just modify that library to remove the traces. But it's a pity to lose the traces of course...

jfjlaros, 2021-12-20:
Can that particular library be configured to use interface other than HardwareSerial?

I ask because it may be possible to use something like [PacketSerial](https://github.com/bakercp/PacketSerial) to multiplex two connections over the same serial line.

cdupont, 2021-12-20:
That library simple outputs traces with `Serial.print`.
Do I need to change it?

jfjlaros, 2021-12-20:
Not for the moment, I have to figure out whether this multiplexing idea is feasible first.

cdupont, 2021-12-23:
So do you think multiplexing is feasible?
With all traces disabled, it's very hard to debug my system now :)

jfjlaros, 2021-12-23:
Yes, I think so.

You might want to check out [this](https://forum.arduino.cc/t/serial-multiplexer/938119/16) post.

Please note that this is all very experimental, it will take some time before we have a stable library that works.

jfjlaros, 2021-12-24:
I have something that seems to be working reasonably, at least on my system.

See the [Arduino library](https://github.com/jfjlaros/serialMux) and the [host library](https://github.com/jfjlaros/arduino-serial-mux) if you want to try it yourself.

cdupont, 2022-03-28:
Hello, I was wondering if you manage to make traces work finally?

jfjlaros, 2022-03-28:
Yes, I have a proof of concept working, I shared the links to this project in a [previous](https://github.com/jfjlaros/simpleRPC/issues/24#issuecomment-1000908916) message.

closed 2022-04-11 by jfjlaros

cdupont, 2022-04-11:
So, I suppose I can use your library serialMux together with simpleRPC and a third library?

```
#include <simpleRPC.h>
#include <serialMux.h>
#include <myOtherLib.h>

SerialMux mux(Serial);
VSerial serialRPC(mux);
VSerial serialTrace(mux);

void setup(void) {
  Serial.begin(9600);
}

void loop(void) {
  //Start RPC lib on multiplexed serial port
  interface(serialRPC, digitalRead, "", digitalWrite, "");
  //Use Serial traces anyway
  serialTrace.printLn("Test");

}
```

jfjlaros, 2022-04-11:
Yes, that is the idea.

Also, if your other library uses `Serial.print()` hardcoded, you can try the [following](https://serialmux.readthedocs.io/en/latest/usage.html#hard-coded-serial-workaround) workaround.

cdupont, 2022-04-11:
That is great, thanks!

jfjlaros, 2022-04-11:
Note that I have only tested the host side on Linux, I don't expect it to work under any other OS, except maybe for macOS.