jfjlaros/simpleRPC #25: Exception/error reporting

cdupont, 2021-12-20:
Hello,
if the function called via RPC throws an exception, what will happen?
In general, how do you advice to report errors in simpleRPC?

cdupont, 2021-12-20:
There are various solutions for error handling... Some functions uses the return value for error codes and uses an argument reference to pass the result. Some others return NULL or -1 for errors. Both solutions are ugly in my opinion :)

cdupont, 2021-12-20:
So, at the moment I don't see any necessary modification...

jfjlaros, 2021-12-20:
I was actually thinking about setting an error bit on the response and depending on this bit, either the result or an error string follows.

This will require a change in the RPC protocol, as well as in the client implementation. There are a couple of other things I want to change in the protocol, so perhaps this could be added to the list.

cdupont, 2021-12-21:
It would be nice to return a sort of sum type: Error | Result
Is it possible right now? Maybe with Object? I didn't quite much understood how to use the Object type.

jfjlaros, 2021-12-21:
Maybe something like this? I can't test it myself at the moment.

```c++
Object<uint8_t, int> fWithException(int a, int b) {
  Object<uint8_t, int> r = {0, 0};  // Error code, return value.

  try {
    get<1>(r) = a / b;
  }
  catch (...) {
    get<0>(r) = 1;
  }
  return r;
}
```

cdupont, 2021-12-21:
Thanks a lot, it works.

jfjlaros, 2021-12-21:
Perhaps this is the way to go then.

I noticed that an extra compiler option is required to enable exception handling (`-fexceptions`), so adding exception handling in the library itself will make it less user friendly.

It is a nice use case to describe in the documentation though.

cdupont, 2021-12-21:
Yes I'm now using `Object<uint8_t, char *>` as a result structure. The first element holds an error code (0 = success) and the second element is my returned value.
I read it in python like that:
```
(r, val) = interface.foo(arg)
match r:
        case 0:
            print(val)
        case 1:
            print("Error: XXX")
        case 2: ...
```

closed 2022-06-01 by jfjlaros