jfjlaros/simpleRPC #26: Java client for Android

aleksas, 2021-12-21:
**Is your feature request related to a problem? Please describe.**
Is there by any chance Java implementation of the client? The goal is to communicate with Android smartphone.

jfjlaros, 2021-12-21:
The Python implementation is meant as a reference implementation, I was actually hoping that other enthusiasts would pick this up for other programming languages. I am not a good Java programmer myself, nor do I have access to an Android OS, so I guess that I am not the best person to ask.

After a bit of searching, I did find that Python is supported on Android. Would that not be a way to go?

aleksas, 2021-12-21:
I've started checking python on Android but it seams a bit backwards. Another alternative might be using [cpp client](https://github.com/jfjlaros/cpp-simple-rpc) and replacing [BT read/write calls](https://github.com/jfjlaros/cpp-simple-rpc/blob/master/src/interface.tcc)  with android bt calls via Android NDK.

I haven't coded for Android in a while now so we'll see how it goes.

jfjlaros, 2021-12-22:
The C++ client is really only a proof of concept for an [other project](https://github.com/SmingHub/Sming/pull/2305) it is not usable as a stand alone library as is.

Perhaps it could be used as a starting point for your project. Please let me know if you have questions.

aleksas, 2022-01-18:
Started [Java port](https://github.com/aleksas/java-simple-rpc) at this moment it's only ported Protocol class and tests from python client.

jfjlaros, 2022-01-18:
That is looking very promising. Nice work.

closed 2022-04-11 by jfjlaros

aleksas, 2022-05-14:
@jfjlaros [arduino-simple-rpc for Java](https://github.com/aleksas/arduino-java-simple-rpc) is mostly done. Code and tests have been mirrored from [arduino-simple-rpc Python package](https://github.com/jfjlaros/arduino-simple-rpc). Package is accessible via Maven central repo. I'll add README and extra tests for more native types eventually, but for now I'll jump to using it in Android project. Would you consider taking over the project?

jfjlaros, 2022-05-16:
That looks pretty good. Congratulations. I will add references to your implementation in the documentation.

> Would you consider taking over the project?

Since I have virtually zero experience with Java projects, I would make a rather poor maintainer for this project. What I can do though, is suggest changes. Probably even in the form of pull requests. Would that be an acceptable compromise?

aleksasp, 2022-05-17:
Yes that should work.

aleksas, 2022-08-07:
@jfjlaros Crude [Android project repo](https://github.com/aleksas/arduino-android-simple-rpc) is ready to some extent. It uses [JAVA simpleRPC](https://github.com/aleksas/arduino-java-simple-rpc). This is more a proof of concept. UI calls only functions without params - other disabled. Full implementation would require dynamic UI generation according to exposed method param list. Due to limited experience with android and lack of free time lately code quality is far from perfect.

jfjlaros, 2022-08-10:
I do not have any way to test it myself, but the animated GIF looks very nice.

In the next release of the library I will mention your projects.