jfjlaros/simpleRPC #30: multiple calls to function returning Vector crashes on the 3rd Call

sgawad, 2023-01-24:
Hello,

First thank you for the library and please excuse me but the behavior is driving me nuts and I don't know how to debug it by myself. I am trying to get data back from a color sensor on arduino through simpleRPC.

here is the declaration of the functions:

```
bool as7341setLed(bool enable_led, uint16_t led_current_ma) {
  as7341.setLEDCurrent(led_current_ma);
  as7341.enableLED(enable_led);
  return enable_led;
}

Vector<uint16_t> as7341read(){
  if (!as7341.readAllChannels(readings))
  {
![OK](https://user-images.githubusercontent.com/5367483/214178461-f3574d95-5b6d-468a-810b-b003f1bdd2ea.png)

    //Serial.println("Error reading all channels!");
    for (int i=0;i<12;++i){readings[i]=0;};
  }else{
    for (int i=0;i<12;++i){u2[i]=readings[i];};
  }
  return u2;
}
```

here is my interface definition:
```
  interface(
  Serial,
  as7341setLed, "as7341setLed: Set LED brightness. @enable_led: enable_led. @led_current_ma: led_current_ma. @return: enable_led.",
  as7341read, "as7341read: Read all color sensors. @return: readings.");
```

I am using the Python interface.
What I observe is that the setLed works everytime and returns reliably everytime
For the read function it reads correctly 2 whole vector of 12 ints... but on the 3rd call the python seems to hang before returning the array.
if I press the reset button on the arduino it traces back io.py in the _read_basic() function
with unpack requires a buffer of 4 bytes.

I added some logging info but it doesn't help me much understand the issue.
Here is a screen capture representing the first two calls
[image]
and here is the 3rd one:
[image]

and here is the Log:
```
1:04:18 AM
<   H   B
<   H   B
<   H   B
1:04:20 AM
<   I   ?
1:04:20 AM
False
1:04:23 AM
<   I   ['H']
1:04:24 AM
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
1:04:24 AM
[95, 199, 309, 537, 2437, 206, 967, 1653, 1905, 1138, 2436]
1:04:29 AM
<   I   ['H']
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
<   I   H
1:04:29 AM
[95, 198, 309, 537, 2434, 206, 969, 1655, 1906, 1140, 2441]
1:04:37 AM
<   I   ['H']
1:04:43 AM
---------------------------------------------------------------------------
error                                     Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_28600\2026148949.py in <module>
----> 1 my_list=interface.as7341read()
      2 my_list

<string> in as7341read(self)

~\AppData\Roaming\Python\Python39\site-packages\simple_rpc\simple_rpc.py in call_method(self, name, *args)
    186         # Read return value (if any).
    187         if method['return']['fmt']:
--> 188             return self._read(method['return']['fmt'])
    189         return None
    190 

~\AppData\Roaming\Python\Python39\site-packages\simple_rpc\simple_rpc.py in _read(self, obj_type)
    101         :returns: Return value.
    102         """
--> 103         return read(
    104             self._connection, self.device['endianness'], self.device['size_t'],
    105             obj_type)

~\AppData\Roaming\Python\Python39\site-packages\simple_rpc\io.py in read(stream, endianness, size_t, obj_type)
     83     logging.info(endianness,' ',size_t,' ',obj_type)
     84     if isinstance(obj_type, list):
---> 85         length = _read_basic(stream, endianness, size_t)
     86         logging.info('it is a list')
     87         return [

~\AppData\Roaming\Python\Python39\site-packages\simple_rpc\io.py in _read_basic(stream, endianness, basic_type)
     29     full_type = (endianness + basic_type).encode('utf-8')
     30     logging.info('read full_type: ',full_type)
---> 31     return unpack(full_type, stream.read(calcsize(full_type)))[0]
     32 
     33 

error: unpack requires a buffer of 4 bytes
```

I tried to change the array to a size of one only ... I tried to change the type by casting it to int.
Nothing seems to allow me to retrieve more than 2 arrays from the sensor...
Thank you for your help.

jfjlaros, 2023-01-24:
I think this may be due to a flaw in the implementation of `Vector`. For now, could you try to create a local vector instead of returning a global one? E.g.,

```cpp
Vector<uint16_t> as7341read() {
  Vector u2(12);
  // ...
  return u2;
}
```

sgawad, 2023-01-24:
Thanks so much Jeroen,

That seems to solve the problem!!
Works well now.

Have a great day,

Shady

jfjlaros, 2023-01-24:
I will keep this open as a reminder.

jfjlaros, 2024-10-19:
A better implementation of `Vector` was added in c5a3d67.

closed 2024-10-19 by jfjlaros