jfjlaros/simpleRPC #31: Crash after processing any command on ESP32

prutschman-iv, 2023-09-29:
I've used simpleRPC successfully on an Arduino and a Teensy 4. I'm now trying it on an ESP32 (Adafruit Huzzah32 feather) and running into a crash. Specifically, the board resets after completing any RPC call, including requesting a command list.

Here's a minimal reproduction:
```
#include <simpleRPC.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
  interface(Serial, millis, "uptime: returns uptime in milliseconds");
}
```
If I reset the board, wait 5 seconds, and then call
`simple_rpc call -b 115200 COM6` I get a response like `7427`. If I issue the command back-to-back, I get responses like `2450`, `2427`, `2475` indicating a reset has taken place. (Probably flashing an LED on reset would be a more obvious way of indicating the reset, now that I think about it.)

Software Versions:
simpleRPC 3.2.0
Arduino IDE 2.2.1
esp32 BSP (Espressif's) 2.0.11

jfjlaros, 2023-09-30:
Connecting to the serial port usually resets the device.

You could test this by leaving the serial connection open between calls, e.g.,

```python
>>> from simple_rpc import Interface
>>> interface = Interface('COM6')
>>> interface.uptime()
1234
>>> interface.uptime()
2345
```

prutschman-iv, 2023-09-30:
It was indeed related to the serial connection, thank you! Persisting the connection, as you suggested, prevents the reboots, so I'm good to go. (Although weirdly, it was *definitely* closing the serial connection that caused the reboot, rather than connecting the serial. I can get the first call to uptime to return an arbitrarily high value by waiting a long time then issuing the first call.)

closed 2023-10-01 by jfjlaros

jfjlaros, 2023-10-02:
Hmm, maybe that is an ESP specific thing then.