jfjlaros/simpleRPC #41: Poor performance over WiFi on ESP32

zshivers, 2024-07-03:
It appears to take about ~300 ms to call one RPC over WiFi. I'm using an [Adafruit ESP32-S2 TFT Feather](https://www.adafruit.com/product/5300) board. 

Arduino code:
```c++
#include <WiFi.h>
#include <simpleRPC.h>

#define SSID "..."
#define PASS "..."
#define PORT 1025

WiFiServer server(PORT);

void setup() {
  WiFi.begin(SSID, PASS);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }

  server.begin();
}

void loop() {
  WiFiClient client{ server.available() };

  if (client) {
    while (client.connected()) {
      interface(
        client,
        millis, F("millis: Get time in milliseconds. @return: Timestamp in milliseconds"),
    }
    client.stop();
  }
}
```

Using a quick script to query `millis()` continuously:
```python
from simple_rpc import Interface
with Interface('socket://REPLACE_IP:1025') as interface:
    last_millis = 0
    while True:
        millis = interface.millis()
        if last_millis > 0:
            print(millis - last_millis)
        last_millis = millis
```
results in approximately 300 ms between timestamps from the Arduino:
```
% python3 rpc_test.py
322
318
321
347
345
328
316
326
327
324
346
...
```

Trying something similar using `Serial` instead of `WiFiClient` as the `Stream` interface for the RPC results 0-1 ms between calls. Is there something that can be done to improve the latency?

jfjlaros, 2024-07-03:
I think this may be related to [this](https://github.com/espressif/arduino-esp32/issues/4882#issuecomment-1148869000) issue. Perhaps you can try the suggestion made there?

jfjlaros, 2024-07-13:
Did this solve your problem?

zshivers, 2024-07-15:
No. Adding `client.setNoDelay()` has no effect on the Arduino side. Using your suggestion on the computer side would require altering the library. Maybe I misinterpreted the required steps in that other bug - any other specific suggestions?

jfjlaros, 2024-07-16:
The problem is on the computer side, at least it was when I ran into this issue.

Apparently, short messages are bundled in one package for efficiency reasons. This may be beneficial for high throughput applications, but not for low volume, low latency applications like the ones we have.

Which library do you need to alter? If it is the `arduino-simple-rpc` Python library, then I can look into it.

I also found [this note](https://github.com/jfjlaros/arduino-simple-rpc/blame/master/examples/wifi/socket_speed_patch.py) I made a while ago. Perhaps you can try to add this to the top of your script?

jfjlaros, 2024-07-27:
The following modification gives a significant speed gain.

```python
from serial.urlhandler import protocol_socket
protocol_socket.time.sleep = lambda x: None

from simple_rpc import Interface
with Interface('socket://REPLACE_IP:1025') as interface:
    last_millis = 0
    while True:
        millis = interface.millis()
        if last_millis > 0:
            print(millis - last_millis)
        last_millis = millis
```

Example:

```bash
$ python test.py 
71
55
66
83
56
153
56
57
56
57
58
71
...
```

jfjlaros, 2024-08-01:
I will close this issue for now. Please feel free to reopen it if the problem persists.

closed 2024-08-01 by jfjlaros